PINTO0309/MobileNet-SSD-RealSense #3: Work with RaspberryPi3, step three: Update udev rule

wo164756189, 2018-08-21:
It occurs an error when I using the following commands with my raspberry pi:
$ sudo apt install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev at-spi2-core libdrm*
###########################################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libdrm-radeon1-dbg' for glob 'libdrm*'
Note, selecting 'libdrm-freedreno1' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau2' for glob 'libdrm*'
Note, selecting 'libdrm2-dbg' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1a' for glob 'libdrm*'
Note, selecting 'libdrm-amdgpu1' for glob 'libdrm*'
Note, selecting 'libdrm-omap1' for glob 'libdrm*'
Note, selecting 'libdrm-omap1-dbg' for glob 'libdrm*'
Note, selecting 'libdrm2' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1a-dbg' for glob 'libdrm*'
Note, selecting 'libdrmaa1.0' for glob 'libdrm*'
Note, selecting 'libdrmaa-dev' for glob 'libdrm*'
Note, selecting 'libdrm-radeon1' for glob 'libdrm*'
Note, selecting 'libdrm-dev' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1-dbg' for glob 'libdrm*'
E: Unable to locate package libglfw3-dev
###########################################

what's the problems?

wo164756189, 2018-08-21:
Thanks. The OS of my pi is Raspbian Stretch.
I have followed your comment. When I run the commands: 
$ sudo apt install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev at-spi2-core libdrm*
Another error occurs.
#########################################################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libdrm-radeon1-dbg' for glob 'libdrm*'
Note, selecting 'libdrm-freedreno1' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau2' for glob 'libdrm*'
Note, selecting 'libdrm2-dbg' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1a' for glob 'libdrm*'
Note, selecting 'libdrm-amdgpu1' for glob 'libdrm*'
Note, selecting 'libdrm-omap1' for glob 'libdrm*'
Note, selecting 'libdrm-omap1-dbg' for glob 'libdrm*'
Note, selecting 'libdrm2' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1a-dbg' for glob 'libdrm*'
Note, selecting 'libdrmaa1.0' for glob 'libdrm*'
Note, selecting 'libdrmaa-dev' for glob 'libdrm*'
Note, selecting 'libdrm-radeon1' for glob 'libdrm*'
Note, selecting 'libdrm-dev' for glob 'libdrm*'
Note, selecting 'libdrm-nouveau1-dbg' for glob 'libdrm*'
libgtk-3-dev is already the newest version (3.22.11-1+rpi3).
libdrm-amdgpu1 is already the newest version (2.4.74-1).
libdrm-amdgpu1 set to manually installed.
libdrm-freedreno1 is already the newest version (2.4.74-1).
libdrm-freedreno1 set to manually installed.
libdrm2 is already the newest version (2.4.74-1).
pkg-config is already the newest version (0.29-4).
libdrm-radeon1 is already the newest version (2.4.74-1).
libdrm-radeon1 set to manually installed.
libdrm-nouveau2 is already the newest version (2.4.74-1).
libdrm-nouveau2 set to manually installed.
git is already the newest version (1:2.11.0-3+deb9u2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libdrm-dev : Depends: libdrm2 (= 2.4.40-1~deb7u2) but 2.4.74-1 is to be installed
              Depends: libdrm-radeon1 (= 2.4.40-1~deb7u2) but 2.4.74-1 is to be installed
 libdrm-radeon1-dbg : Depends: libdrm-radeon1 (= 2.4.40-1~deb7u2) but 2.4.74-1 is to be installed
 libdrm2-dbg : Depends: libdrm2 (= 2.4.40-1~deb7u2) but 2.4.74-1 is to be installed
 libusb-1.0-0-dev : Depends: libusb-1.0-0 (= 2:1.0.11-1) but 2:1.0.21-1 is to be installed
E: Unable to correct problems, you have held broken packages.
#######################################################################

PINTO0309, 2018-08-21:
@wo164756189 
It seems that the procedure of Ubuntu 16.04 was mixed.
I am sorry many times but can you try the following?
```
$ sudo apt install -y git libssl-dev pkg-config libgtk-3-dev at-spi2-core
```
Probably the following procedure is unnecessary.
```
$ cd ~
$ git clone https://github.com/glfw/glfw
$ cd glfw
$ mkdir build;cd build
$ cmake ..
$ make -j1
$ sudo make install
```

wo164756189, 2018-08-21:
Thanks for your timely reply. I have tried.
The results as follows:
##########################################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'mesa-utils' for glob 'mesa-utils*'
Note, selecting 'libglu1-xorg-dev' for glob 'libglu1*'
Note, selecting 'libglu1-mesa-dev' for glob 'libglu1*'
Note, selecting 'libglu1' for glob 'libglu1*'
Note, selecting 'libglu1-mesa' for glob 'libglu1*'
Note, selecting 'libglu1-mesa' instead of 'libglu1'
libglu1-mesa is already the newest version (9.0.0-2.1).
libglu1-mesa set to manually installed.
mesa-utils is already the newest version (8.3.0-3).
mesa-utils set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgles2-mesa-dev : Depends: libegl1-mesa-dev but it is not going to be installed
 libglu1-mesa-dev : Depends: libglu1-mesa (= 8.0.5-4+deb7u2+rpi1) but 9.0.0-2.1 is to be installed
                    Depends: libgl1-mesa-dev but it is not going to be installed or
                             libgl-dev
 libopenal-dev : Depends: libopenal1 (= 1:1.14-4) but 1:1.17.2-4 is to be installed
E: Unable to correct problems, you have held broken packages.
######################################################################

PINTO0309, 2018-08-21:
@wo164756189 
After all, this may have been enough...
```
$ sudo apt install -y git pkg-config at-spi2-core
```
If your problem is fixed, I will review the implementation procedure immediately.

wo164756189, 2018-08-21:
@PINTO0309 
I have tried the commands: `$ sudo apt install -y git pkg-config at-spi2-core`
It's OK now, but these dependencies seems not enough to compile RealSense SDK.

When I follow the step 10. Install Intel® RealSense™ SDK 2.0 with the commands:
$ sudo apt install mesa-utils* libglu1* libgles2-mesa-dev libopenal-dev
It occurs errors:
############################################################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'mesa-utils' for glob 'mesa-utils*'
Note, selecting 'libglu1-xorg-dev' for glob 'libglu1*'
Note, selecting 'libglu1-mesa-dev' for glob 'libglu1*'
Note, selecting 'libglu1' for glob 'libglu1*'
Note, selecting 'libglu1-mesa' for glob 'libglu1*'
Note, selecting 'libglu1-mesa' instead of 'libglu1'
libglu1-mesa is already the newest version (9.0.0-2.1).
libglu1-mesa set to manually installed.
mesa-utils is already the newest version (8.3.0-3).
mesa-utils set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgles2-mesa-dev : Depends: libegl1-mesa-dev but it is not going to be installed
 libglu1-mesa-dev : Depends: libglu1-mesa (= 8.0.5-4+deb7u2+rpi1) but 9.0.0-2.1 is to be installed
                    Depends: libgl1-mesa-dev but it is not going to be installed or
                             libgl-dev
 libopenal-dev : Depends: libopenal1 (= 1:1.14-4) but 1:1.17.2-4 is to be installed
E: Unable to correct problems, you have held broken packages.
#########################################################################
I have ignored this error, and continue the commands:
$ cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release
Another error (only the error part as follows): 
#########################################################################
CMake Error at /home/pi/cmake-3.11.4/share/cmake-3.11/Modules/FindPkgConfig.cmake:640 (message):
  None of the required 'libusb-1.0' found
Call Stack (most recent call first):
  CMakeLists.txt:738 (pkg_search_module)

CMake Error at CMakeLists.txt:744 (message):
  Failed to find libusb-1.0

-- Configuring incomplete, errors occurred!
See also "/home/pi/librealsense/build/CMakeFiles/CMakeOutput.log".
###########################################################################
To solve this issue, I used the commands: $ sudo apt-get install libusb-1.0.0-dev
but fails as follows:
###########################################################################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0.0-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libusb-1.0-0-dev : Depends: libusb-1.0-0 (= 2:1.0.11-1) but 2:1.0.21-1 is to be installed
E: Unable to correct problems, you have held broken packages.
###########################################################################

What should I do to install RealSense SDK2.0?

PINTO0309, 2018-08-21:
@wo164756189 

>What should I do to install RealSense SDK2.0?

```
$ sudo apt remove libusb-1.0-0 libusb-1.0-0-dev
$ sudo apt install libusb-1.0-0 libusb-1.0-0-dev
```

When using a USB camera, although it is not required, you need to rework the program a bit.

1. Please start work from "15.Installing the OpenGL package for Python".
2. Comment out all the parts to read RealSense module.

I can not cover everything, but for example ...

**MultiStickSSDwithRealSense.py**
```
#import pyrealsense2 as rs
```
```
#        pipeline = rs.pipeline()
#        config = rs.config()
#        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
#        config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
#        profile = pipeline.start(config)
#        depth_sensor = profile.get_device().first_depth_sensor()
#        depth_scale = depth_sensor.get_depth_scale()
#        align_to = rs.stream.color
#        align = rs.align(align_to)
#        window_name = "RealSense"
```
```
#            # Wait for a coherent pair of frames: depth and color
#            frames = pipeline.wait_for_frames()
#            depth_frame = frames.get_depth_frame()
#            color_frame = frames.get_color_frame()
#            if not depth_frame or not color_frame:
#                continue
#            if frameBuffer.full():
#                frameBuffer.get()
#            color_image = np.asanyarray(color_frame.get_data())
```
```
#            # 0:= No background transparent, 1:= Background transparent
#            if background_transparent_mode == 0:
#                depth_frame = frames.get_depth_frame()
#                color_frame = frames.get_color_frame()
#
#            elif background_transparent_mode == 1:
#                aligned_frames = align.process(frames)
#                depth_frame = aligned_frames.get_depth_frame()
#                color_frame = aligned_frames.get_color_frame()
#
#            depth_dist  = depth_frame.as_depth_frame()
#            depth_image = np.asanyarray(depth_frame.get_data())
#            color_image = np.asanyarray(color_frame.get_data())
```
```
#                    meters = depth_dist.get_distance(box_left+int((box_right-box_left)/2), box_top+int((box_bottom-box_top)/2))
#                    label_text = LABELS[int(class_id)] + " (" + str(percentage) + "%)"+ " {:.2f}".format(meters) + " meters away"
```

wo164756189, 2018-08-22:
@PINTO0309 
Thanks, all the previous problems are solved! But there is a new problem.
When I finished Step 15.Installing the OpenGL package for Python, and back to Step 10.Install Intel® RealSense™ SDK 2.0. A weird error occurs when I execute:
`$ cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release`

The info of the error part as follows:
`-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for one of the modules 'libusb-1.0'
-- Could NOT find OpenGL (missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY OPENGL_INCLUDE_DIR) 
CMake Error at examples/CMakeLists.txt:23 (message):
  

  

   OpenGL package is missing!

  

-- Configuring incomplete, errors occurred!
See also "/home/pi/librealsense/build/CMakeFiles/CMakeOutput.log".
`

PINTO0309, 2018-08-22:
@wo164756189 
Please try below.
```
$ sudo apt install libgl1-mesa-dev
```

wo164756189, 2018-08-22:
@PINTO0309
$ sudo apt install libgl1-mesa-dev
fails:
#######################################################
The following packages have unmet dependencies:
 libgl1-mesa-dev : Depends: mesa-common-dev (= 13.0.6-1+rpi2) but it is not going to be installed
                   Depends: libdrm-dev (>= 2.4.69) but it is not going to be installed
                   Depends: libx11-xcb-dev but it is not going to be installed
                   Depends: libxcb-dri3-dev but it is not installable
                   Depends: libxcb-present-dev but it is not installable
                   Depends: libxcb-sync-dev but it is not installable
                   Depends: libxshmfence-dev but it is not installable
                   Depends: libxcb-dri2-0-dev but it is not going to be installed
                   Depends: libxcb-glx0-dev but it is not going to be installed
                   Depends: libxxf86vm-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

PINTO0309, 2018-08-22:
@wo164756189 
I do not understand why such an error occurs.
Let's install one missing dependency according to the error message.
```
$ sudo apt update;sudo apt upgrade
$ sudo apt install mesa-common-dev libdrm-dev libx11-xcb-dev libxcb-dri3-dev libxcb-present-dev libxcb-sync-dev libxshmfence-dev libxcb-dri2-0-dev libxcb-glx0-dev libxxf86vm-dev libgl1-mesa-dev
```

wo164756189, 2018-08-22:
@PINTO0309 
I have tried your update in step 3 with the commands:
# Raspbian Stretch
$ sudo apt install -y git libusb-1.0.0 libusb-1.0-0-dev pkg-config at-spi2-core libgtk-3-dev \
mesa-common-dev libdrm-dev libx11-xcb-dev libxcb-dri3-dev libxcb-present-dev libxcb-sync-dev \
libxshmfence-dev libxcb-dri2-0-dev libxcb-glx0-dev libxxf86vm-dev libgl1-mesa-dev

But there are still some problems:
##########################################################
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0.0'
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0.0'
Package libxcb-dri3-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libxcb-present-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libxcb-sync-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libxshmfence-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libxcb-dri3-dev' has no installation candidate
E: Package 'libxcb-present-dev' has no installation candidate
E: Package 'libxcb-sync-dev' has no installation candidate
E: Package 'libxshmfence-dev' has no installation candidate
###################################################################

One more question: Did you setup the environment for both Respbian and Ubuntu Mate on Respberry Pi for this project successfully?

PINTO0309, 2018-08-22:
@wo164756189 
>One more question: Did you setup the environment for both Respbian and Ubuntu Mate on Respberry Pi for this project successfully?

Yes. **I worked with the initial image "Raspbian Stretch".(No Ubuntu Mate)**
**https://downloads.raspberrypi.org/raspbian_latest**

All the videos released on Youtube are those of "Raspbian Stretch" and "Ubuntu 16.04".
In addition, it performed in almost the same procedure as described in this repository.
I am puzzled as much as problems arise.

There are Japanese who have succeeded quoting my procedure.
https://sekailab.com/wp/2018/08/14/librealsense-with-armbian-on-orangepi-pc2/

Strangely, in your environment, the 'stable' package of the official site will be an error.
By the way, the official procedure of Intel is below, but it fails even if it is executed as instructed.
"Ubuntu Mate"
https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

Although it is described in my blog, I ran about three times and have not failed so far.
https://qiita.com/PINTO/items/2ad10526f9b2e1c8cdf3

Something else may be necessary...

The only option I can think of is trying the following.
```
$ cd ~
$ wget http://ftp.jp.debian.org/debian/pool/main/libx/libxcb/libxcb-dri3-dev_1.12-1_armhf.deb
or
$ wget http://ftp.cn.debian.org/debian/pool/main/libx/libxcb/libxcb-dri3-dev_1.12-1_armhf.deb

$ sudo dpkg -i libxcb-dri3-dev_1.12-1_armhf.deb

$ wget http://ftp.jp.debian.org/debian/pool/main/m/mesa/libgl1-mesa-dev_13.0.6-1+b2_armhf.deb
or
$ wget http://ftp.cn.debian.org/debian/pool/main/m/mesa/libgl1-mesa-dev_13.0.6-1+b2_armhf.deb

$ sudo dpkg -i libgl1-mesa-dev_13.0.6-1+b2_armhf.deb
```
etc...

PINTO0309, 2018-08-22:
@wo164756189 
I will propose the last means as long as I can think of.

I will give you important questions.
Are you Chinese?
If correct, please try the following at the end.
```
$ sudo nano /etc/apt/sources.list

deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
↓modify
deb http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian/ stretch main contrib non-free rpi

$ sudo apt update;sudo apt upgrade
```
If it is incorrect, please select the URL closest to your country from the following website.
http://raspbian.org/RaspbianMirrors

wo164756189, 2018-08-23:
@PINTO0309 
That's the point! I edited the "source.list" before, and it looks like this before:
#deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
deb http://mirrors.aliyun.com/raspbian/raspbian/ wheezy main non-free contrib
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
deb-src http://mirrors.aliyun.com/raspbian/raspbian/ wheezy main non-free contrib

I used the source from aliyun, and uncomment the last line for deb-src. I don't know whether the problem is the source link or the last line(I uncomment it)?

Whatever, it works now and the new "source.list" looks like this as your advised:
#deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
deb http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi

In the third step, I think the commands should be like this for Raspbian (the same as Ubuntu):
$ sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
$ sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev 

I have also succeed in Raspberry Pi for UbuntuMate 16.04. It is similar to PC based Ubuntu 16.04 (as you described). But you will have to add the swap file in Raspberry Pi for UbuntuMate 16.04 just like in Raspbian.

PINTO0309, 2018-08-23:
@wo164756189 

>I used the source from aliyun, and uncomment the last line for deb-src. I don't know whether the problem is the source link or the last line(I uncomment it)?

OK. My guess was correct.
Below is quite a problem.
**"wheezy"**
```
deb http://mirrors.aliyun.com/raspbian/raspbian/ wheezy main non-free contrib
deb-src http://mirrors.aliyun.com/raspbian/raspbian/ wheezy main non-free contrib
```
Originally the following is the correct description.
```
deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
deb-src http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
```

>I think the commands should be like this for Raspbian (the same as Ubuntu):

Thank you!
I'll fix the procedure.

>But you will have to add the swap file in Raspberry Pi for UbuntuMate 16.04 just like in Raspbian.

All right. I understand.
From last night to this morning, I fixed the procedure a little more.
Readme added : "PC + Ubuntu16.04"

Congratulations!!
If all the problems seem to be solved, please close "issue".

PINTO0309, 2018-08-26:
@wo164756189 
It seems that the problem has been solved, so I will close the "issue".
If another problem occurs, please make "issue" once more.
This is to make it an easy-to-read "issue" for other engineers.

closed 2018-08-26 by PINTO0309