PINTO0309/MobileNet-SSD-RealSense #4: NCSDK setup

bsense-rius, 2018-08-23:
Dear Master Pinto,

in your very thorough and kind step by step I miss the NCSDK v2 setup... when do you consider best to install it (after which step number)?

After seeing some stretch issues, do you use ubuntu 16.04 in rasp? which exactly (@ rasp site there are two: ubuntu mate & snappy)

thx!!

PINTO0309, 2018-08-23:
@bsense-rius 

>in your very thorough and kind step by step I miss the NCSDK v2 setup... when do you consider best to install it (after which step number)?

Immediately after "2.Extend the SWAP area (Raspbian Stretch Only)" is the best.

>After seeing some stretch issues, do you use ubuntu 16.04 in rasp? which exactly (@ rasp site there are two: ubuntu mate & snappy)

I think that "Ubuntu Mate" and "Ubuntu 16.04" are essentially the same.
However, I have not tried the configuration of RaspberryPi + Ubuntu 16.04.
The official support OS of each maker is "Ubuntu".
Since the procedure by "Raspbian" is my own procedure, it is informal.

PINTO0309, 2018-08-23:
@bsense-rius 
I fixed the procedure.

bsense-rius, 2018-08-23:
so... what is your setup so that we could reproduce all the steps? raspbian stretch or ubuntu mate? the youtube videos seem that it is raspbian.

anyway thanks!

PINTO0309, 2018-08-23:
@bsense-rius 

**Raspbian Stretch**

#3 However, just as before, "Ubuntu Mate" seems to succeed.

closed 2018-08-23 by bsense-rius