PINTO0309/MobileNet-SSD-RealSense #6: RealSense D435 connection problems

wo164756189, 2018-08-23:
Hi, PINTO0309
I have the environment: Raspberry Pi + Raspbian Stretch+ RealSense D435.
There are some problems with my D435 connecting to my Pi.
 The screenshot as follows:
[image]

1.  I have no RGB module, only stereo module in my $ realsense-viewer
2.  Incomplete frame (38%) and no frames received. (Maybe the transfer speed is not enough with usb2.0? But it is perfect in your demo. )

PINTO0309, 2018-08-23:
@wo164756189 
Have you updated the D435 firmware?
**Firmware Ver 5.9.13**

wo164756189, 2018-08-23:
@PINTO0309 
Not update yet. Do you have the instructions for updating in Raspbian?
Is the pdf document ok? 
https://www.intel.com/content/www/us/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html

PINTO0309, 2018-08-23:
@wo164756189 
It's written in the Readme.
You seem to have misunderstood.
A "PC" different from RaspberryPi is required.

Windows PC or Ubuntu PC

◆Windows
https://downloadcenter.intel.com/download/27514/Windows-Device-Firmware-Update-Tool-for-Intel-RealSense-D400-Product-Family

◆Ubuntu
https://www.intel.com/content/www/us/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html

wo164756189, 2018-08-24:
Now the Firmware is updated to Ver 5.9.14
And I encountered another problem as the screenshot.
$ sudo raspi-config
"7.Advanced Options" - "A7 GL Driver" - "G2 GL (Fake KMS)"
$ realsense-viewer
[image]
The RGB image is black, and I use command "$ rs-capture" to prove this.
[image]
This means the RGB frame can be received, but all the bytes are 00 00 00 00 .....
So how can I get the real RGB frame?

PINTO0309, 2018-08-24:
@wo164756189 
I am not doing anything special.
Please try Firmware Ver 5.9.13.
Firmware Ver 5.9.14 is development release.
Firmware Ver 5.9.13 is production release.
https://downloadmirror.intel.com/27924/eng/Intel%C2%AE%20RealSense%E2%84%A2D400%20Series%20Signed%20Production%20Firmware%20v5_9_13.zip

PINTO0309, 2018-08-26:
@wo164756189 
Congratulations!
Recent big manufacturers release software without barely testing.
Experience, the latest version can not be trusted so much.

PINTO0309, 2018-08-26:
@wo164756189 
It seems that the problem has been solved, so I will close the "issue".
If another problem occurs, please make "issue" once more.
This is to make it an easy-to-read "issue" for other engineers.

closed 2018-08-26 by PINTO0309