PINTO0309/MobileNet-SSD-RealSense #8: ncsdk "make examples" - installs OpenCV (is it really needed?)

bonastreyair, 2018-08-23:
Dear PINTO, 

I noted that when doing "make examples -j1" when installing ncsdk it automatically installs OpenCV...
I understand that if you are doing a fresh install there is no need to install it to later uninstall it and then install it again in your step-by-step, so do I really need to make the examples in this step or I can psotpone this step after the whole step-by-step process?

Thanks in avdance!

btw: you made an amazing work! thks a lot!

PINTO0309, 2018-08-23:
@bonastreyair 

>so do I really need to make the examples in this step or I can psotpone this step after the whole step-by-step process?

"make examples" is not necessary.
It does not matter if it is executed at the end.
However, I recommend reinstalling tensorflow.
Also, protobuf (>=3.5.1) is required to install tensorflow (>=1.8.0).
Tensorflow1.7.0 is very unstable.

>I also see now there are repeated steps regarding the installation of protobuf 3.5.1, do I really need to install it twice?

Sorry. It is a mistake in correcting the procedure.
Because I corrected it in a hurry, I forgot to delete one.
I will correct it immediately.

bonastreyair, 2018-08-23:
Thanks a lot! 

Issue closed!

closed 2018-08-23 by bonastreyair