PINTO0309/MobileNet-SSD-RealSense #9: Systemdでの起動失敗 Automatic startup failure on Systemd

Teable, 2018-10-09:
pyrealsense2を用いたプログラムをsystemdを用いて起動を試みたのですが(python3)、以下のエラーにより失敗します。
シェル上でpython3起動した際は、"ImportError"は出ず、プログラムも正常に動作します。

ImportError: No module named 'pyrealsense2'

パッケージのバージョンは全てreadmeのままです。

宜しくお願い致します。

English
I attempted to run a program using pyrealsense2 with systemd (python 3), but it fails with the following error.

ImportError: No module named 'pyrealsense2

All package versions compliant with  readme.

Thank you.

PINTO0309, 2018-10-09:
@Teable 
◆Japanese
pyrealsense2 に限った事象ではないように見えますが、下記は試されましたか？
と、systemdを使用したことない私が語ります。
https://bacchi.me/linux/systemd-tips/

◆English
It seems that it is not limited to RealSense, but has the following been tried?
https://bacchi.me/linux/systemd-tips/

Teable, 2018-10-09:
import pyrealsense2をコメントアウトしたところ、systemdにて正常に起動したので、おそらくpyrealsense2に問題があるかと...

PINTO0309, 2018-10-09:
@Teable 
注目していただきたいのは下記の手順の箇所です。

13.Update PYTHON_PATH
```
$ nano ~/.bashrc
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
```
.bashrc が実行される前に systemd が実行されている、と予想しました。
根拠は、エラーメッセージが、 "pyrealsense2を見つけられない" という表示だったためです。

先人の知恵によると。。。
１．　/etc/sysconfig/USER_NAME に読み込ませたい環境変数を書く
２．　systemd設定ファイルの[service]ディレクティブにEnvironmentFileを定義する
３．　サービスの再起動

で解決するようです。

ちなみに、私の環境で pyrealsense2 の実体がどこにあるか調べてみました。
```
$ sudo find / -name pyrealsense2*

/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2.15.0 ←ココ
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2 ←ココ
/usr/local/lib/pyrealsense2.cpython-35m-x86_64-linux-gnu.so ←ココ
/home/xxxxx/librealsense/wrappers/python/pyrealsense2
/home/xxxxx/librealsense/build/wrappers/python/pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2.15.0
/home/xxxxx/librealsense/build/wrappers/python/CMakeFiles/pyrealsense2.dir
/home/xxxxx/librealsense/build/wrappers/python/pyrealsense2.cpython-35m-x86_64-linux-gnu.so.2
/home/xxxxx/librealsense/build/wrappers/python/pyrealsense2.cpython-35m-x86_64-linux-gnu.so
```
どうでしょうか。

Teable, 2018-10-09:
ご丁寧にありがとうございます。
なるほど、分かりました。
確認してみます。

Teable, 2018-10-10:
正常に動作しました。ありがとうございます!

修正点:
sudo nano /etc/sysconfig/USER_NAME 

以下の行を追加:
PYTHONPATH=$PYTHONPATH:/usr/local/lib
(PATHはpyrealsense2がある場所を指定。Readmeに沿って環境構築した場合は、これで大丈夫です。)

後は上記の方法に従い、解決しました。

◆English
Service worked normally. Thank you!

modified method:
sudo nano /etc/sysconfig/USER_NAME 

Add the following line:
PYTHONPATH=$PYTHONPATH:/usr/local/lib
(In PATH, Specify address where pyrealsense 2 exists. If you built environment according to Readme, it is "/usr/local/lib" .)

Afterwards, if you follow the above method, This error is solved.

Teable, 2018-10-10:
解決致しましたので、issueを閉じさせて頂きます。

closed 2018-10-10 by Teable