PINTO0309/MobileNet-SSD-RealSense #10: Feedback

nunux13, 2018-12-31:
Hello,

Many thanks for your code.

Got about 15 FPS, using USB Webcam on Rpi 3 B+ and 2 ncs2.
Have you bench using 4 ncs on rpi ?

I use https://github.com/umlaeute/v4l2loopback to feed trafic cam view.

I get better result starting cam in 640x480 and cropping like this:

color_image=cv2.resize(color_image,(532,400))
color_image=color_image[100:100+300,116:116+300]

I also try to use the script with yolov3 but got buffer overflow.

Regards,

Pierre

PINTO0309, 2018-12-31:
@nunux13 

Thank you for giving feedback.
I am just implementing MultiProcess + YoloV3.
And thank you for providing very useful information.

>Have you bench using 4 ncs on rpi ?

I plan to give it a try as soon as I finish implementing MultiProcess + YoloV3.

btw, Japan is a new year in just 2 hours.

PINTO0309, 2019-01-01:
@nunux13 

NCS2 x4 + RaspberryPi3 ---> 24 FPS
The ARM architecture CPU does not seem to be able to demonstrate sufficient performance.

By the way, implementation of YoloV3 's MultiProcess is also over.
**https://github.com/PINTO0309/OpenVINO-YoloV3.git**
YoloV3 + NCS2 x4 + RaspberryPi3 ---> 1.7 FPS

nunux13, 2019-01-02:
@PINTO0309 

May you achieve the goals of your life this year. Happy New Year!!

NCS2 x2 + ODroid XU4Q on Debian ----> 20 FPS

ODroid XU4Q is ARM based with USB3 ports so 
I bet we get USB2 throttling on Raspberry 3B+

I will order 1 more stick to reach my goal of 30 FPS

Regards,

PINTO0309, 2019-01-04:
@nunux13

Only performance tuning is over.
MultiStickSSDwithRealSense_OpenVINO_NCS2.py

- Youtube (RaspberryPi3, NCS2 x1, 24 FPS)
https://youtu.be/Xj2rw_5GwlI

- GIF
[image]

nunux13, 2019-01-08:
Wonderful !