PINTO0309/MobileNet-SSD-RealSense #11: NCS2 Slower Than NCS1

Luxonis-Brandon, 2019-01-03:
Hi PINTO0309,

Thanks for all your work here and particularly for documenting it all.  I've bene able to successfully follow everything, which I think speaks to your thoroughness.  :-)

So I've setup and run, with RealSense D435:

1. No NCS with TensorFlow ([here](https://github.com/PINTO0309/MobileNet-SSDLite-RealSense-TF)
2. NCS1 (this repository, with NCSDK)
3. NCS2 (this repository, with OpenVINO), using `python3 MultiStickSSDwithRealSense_OpenVINO_NCS2.py`

So I'm seeing the following framerates:
1. 1 FPS
2. 6 FPS
3. 5 FPS

So I'm curious to see that the NCS2 with OpenVINO actually seems to be slightly slower than the NCS1 with NCSDK.

Now my question is, it feels like I'm running the wrong command for 3... am I?

I'm running `python3 MultiStickSSDwithRealSense_OpenVINO_NCS2.py` and the only other options I see there are:

- `MultiStickSSDwithRealSense.py`
- `SingleStickSSDwithRealSense.py`
- `SingleStickSSDwithUSBCamera_OpenVINO_NCS2.py`

So it kind of makes me think that `SingleStickSSDwithRealSense_OpenVINO_NCS2.py`, or something like that, is missing from github (like forgot to be committed, or similar).

Thoughts?

Thanks again!
-Brandon

PINTO0309, 2019-01-04:
@Luxonis-Brandon 

Thank you for giving feedback.

>So I'm curious to see that the NCS2 with OpenVINO actually seems to be slightly slower than the NCS1 with NCSDK.

Quite a while ago I got the same benchmark as you.
Engineers from other countries also convey the same results.

>So it kind of makes me think that SingleStickSSDwithRealSense_OpenVINO_NCS2.py, or something like that, is missing from github (like forgot to be committed, or similar).

Sorry... I have not created "SingleStickSSDwithRealSense_OpenVINO_NCS2.py".
It is the same as using "MultiStickSSDwithRealSense_OpenVINO_NCS2.py" in NCS2 one piece.
"SingleStickSSDwith***" is simple logic for program verification process.

>Oh and to see if SingleStickSSDwithRealSense.py happened to be compatible with both NCS1 (NCSDK) and NCS2 (OpenVINO), I rant it with the NCS2 connected (and no NCS1 connected), and got:

When you install OpenVINO, environment variables are forcibly updated.
In order to make both coexist, I think that tricky environment setting is necessary.

PINTO0309, 2019-01-04:
@Luxonis-Brandon 

Thank you for providing useful information.
I have skills to use Docker, but I will try to make this repository a docker a little more later.
I'm managing multiple repositories alone, at the same time.
I'd like to try a variety of things that you suggested, but I do not have enough time... :sob: 

>That, or see if maybe the ARM servers from Scaleway (here) are usable.

This is a very interesting service.
I want to try various things. :smiley:

PINTO0309, 2019-01-04:
@Luxonis-Brandon
Only performance tuning is over.
MultiStickSSDwithRealSense_OpenVINO_NCS2.py

- Youtube (NCS2 x1, 24 FPS)
https://youtu.be/Xj2rw_5GwlI

- GIF
[image]

Luxonis-Brandon, 2019-01-04:
Sweet!!!  That is very exciting @PINTO0309 !  Will try this today.  

Thanks again!

Luxonis-Brandon, 2019-01-04:
> @Luxonis-Brandon
> 
> Thank you for providing useful information.
> I have skills to use Docker, but I will try to make this repository a docker a little more later.
> I'm managing multiple repositories alone, at the same time.
> I'd like to try a variety of things that you suggested, but I do not have enough time... 😭
> 
> > That, or see if maybe the ARM servers from Scaleway (here) are usable.
> 
> This is a very interesting service.
> I want to try various things. 😃

Heh, thank you!  You've saved me a ton of time with these repositories; much appreciated.

I'm happy to help as I can as well.  If you want anything offloaded I'm happy to try to do it.  

Thanks again,
Brandon

closed 2019-01-04 by Luxonis-Brandon

Luxonis-Brandon, 2019-01-04:
> @Luxonis-Brandon
> Only performance tuning is over.
> MultiStickSSDwithRealSense_OpenVINO_NCS2.py
> 
> * Youtube (NCS2 x1, 24 FPS)
>   https://youtu.be/Xj2rw_5GwlI
> * GIF
>   [image]

Just wanted to note that I did try this, and it is indeed about 4x faster than before, which is awesome.

I did notice a new bug introduced by the change though, which I noted here: [https://github.com/PINTO0309/MobileNet-SSD-RealSense/issues/13](https://github.com/PINTO0309/MobileNet-SSD-RealSense/issues/13)

PINTO0309, 2019-02-17:
I have given up my registration because I can not enter the phone number.
It is very disappointing...

**Scaleway**
**https://www.scaleway.com/virtual-cloud-servers/#anchor_arm**

Luxonis-Brandon, 2019-02-18:
I could probably sign up and then transfer the account if you want?  Thoughts?

PINTO0309, 2019-02-18:
Thank you as always, Brandon.

>I could probably sign up and then transfer the account if you want?

What should I do?
I am a little worried because I do not know how to pay the money.

Luxonis-Brandon, 2019-02-19:
Ya, I'll have to figure that out as well.  I also have a friend who uses these guys all the time.  I'll reach out to him for advise.