PINTO0309/MobileNet-SSD-RealSense #16: MobileNet-SSD + RPi + NCS2

lbcastro, 2019-01-23:
I've been trying to use this repository to run a custom MobileNet-SSD model trained with tensorflow and converted to IR. I have validated a FP32 version of this model using OpenVino's object_detection_sample_ssd, but I'm using a FP16 version for this repository.

I'm running SingleStickSSDwithUSBCamera_OpenVINO_NCS2.py, modified to load my .xml/.bin model and my labels, but I never get any detections. Any idea why? Thanks!

lbcastro, 2019-01-23:
Also, I can get your code to run with the default model it comes with. It's just when I change to my own model that things start to fail.

lbcastro, 2019-01-23:
This might help, if I run the same model with your MultiStickSSDwithRealSense_OpenVINO_NCS2.py (by the way, amazing FPS on this one!), I get a spam of these errors:

Traceback (most recent call last):
  File "/home/pi/MobileNet-SSD-RealSense/MultiStickSSDwithRealSense_OpenVINO_NCS2.py", line 243, in predict_async
    out = self.exec_net.requests[dev].outputs["detection_out"].flatten()
KeyError: 'detection_out'

lbcastro, 2019-01-23:
Just noticed that your default models were converted from caffe and mine is from tensorflow. I guess this is the origin of the problem.

PINTO0309, 2019-01-23:
@lbcastro

According to official information of NCS/NCS2, Tensorflow's MobileNet-SSD is not supported.
As a matter of fact, I've never tried converting Tensorflow models.
My program currently supports conversion only from the Caffe model.
I will try it if I have enough free time.

If possible, can you share your Tensorflow model (.pb) with me?

lbcastro, 2019-01-24:
Thanks for the answer.

I now have a caffe model but the results are the same. I'm using your MultiStickSSDwithRealSense_OpenVINO_NCS2.py script (btw, very impressive FPS on the default model), but I get no detections. Besides changing the LABELS, self.model_xml and self.model_bin variables, is there anything that I should adapt in order to use a custom model?

Later on I can also try to help you adapt to a frozen TF model, but I guess that this issue is probably more important to you. Do you know if other people were successful in using custom models with your code? Thanks a lot!

lbcastro, 2019-01-24:
A good way for me to start debugging this problem would be if you could share how you converted your caffemodel to IR. This is what I'm doing:

sudo python3 mo.py --data_type=FP16 --input_model=mymodel.caffemodel --input_proto=deploy.prototxt --tensorflow_use_custom_operations_config ssd_support.json --batch 1

I tried both with and without the last two arguments (--tensorflow_use_custom_operations_config, --batch), with the same results.

lbcastro, 2019-01-24:
I just tested converting the caffemodel available in your repository and it works, so the problem is really in my model. Trying now to figure out why.

lbcastro, 2019-01-24:
If I run the sample "object_detection_sample_ssd" available in OpenVino I get different behaviors depending if I'm using your model or mine (both converted to IR).

Yours:
[ INFO ] Processing output blobs
[0,15] element, prob = 0.434433    (-7.14125,26.9382)-(324.561,233.211) batch id : 0
[ INFO ] Image out_0.bmp created!

Mine:
[ INFO ] Processing output blobs
[0,1] element, prob = 0.00348474    (168.137,73.1168)-(250.893,210.556) batch id : 0
[1,1] element, prob = 0.00304914    (162.556,75.1975)-(294.11,220.45) batch id : 0
[2,1] element, prob = 0.00134124    (159.083,76.2569)-(257.979,117.03) batch id : 0
[3,1] element, prob = 0.000717487    (127.367,45.4956)-(280.763,209.201) batch id : 0
[4,1] element, prob = 0.000666454    (146.009,84.7207)-(223.752,206.324) batch id : 0
[5,1] element, prob = 0.000520843    (163.775,84.6187)-(257.716,130.648) batch id : 0
[6,1] element, prob = 0.000351223    (0,16.7114)-(174.604,225.62) batch id : 0
[7,1] element, prob = 0.000342579    (56.9683,64.815)-(255.628,183.466) batch id : 0
[8,1] element, prob = 0.000292516    (6.51987,53.0098)-(295.382,166.172) batch id : 0
[9,1] element, prob = 0.000259259    (147.067,63.467)-(269.583,185.277) batch id : 0
[10,1] element, prob = 0.000239865    (169.249,97.7866)-(307.788,183.419) batch id : 0
...
...
[55,4] element, prob = 0.807157    (81.5983,169.039)-(139.829,235.278) batch id : 0 WILL BE PRINTED!
...
...
[ INFO ] Image out_0.bmp created!

I hope this is a good clue for the problem.
(Sorry about posting a lot, but I'm trying to provide as much information as possible)
Thanks!

lbcastro, 2019-01-24:
Thanks for taking the time to help.

Does that explanation mean that I need to include some training parameters in your scripts? Do I need to modify, for instance, any variables in MultiStickSSDwithRealSense_OpenVINO_NCS2.py (besides the location of my model and my labels) related to my training process?

lbcastro, 2019-01-25:
Thanks a lot for all that information, I'll see if it solves my problem.

lbcastro, 2019-01-28:
I still didn't manage to get your repo working with a custom dataset. I'm doing the training process with the VOC datasets as described on your readme file to see if I can get the resulting weights working on the NCS2.

lbcastro, 2019-01-28:
I think I got it. The problem seems to be that with FP16 the loss needs to be much lower than with FP32. Given that I was only able to validate my model with the OpenVino samples which only support FP32, everything seemed fine until I loaded the model in your code. However, I've waited for a while longer and I'm starting to get some detections using my model with your code.

One thing that really helped was adjusting the learning rate after some iterations. If I set a high learning rate in the beginning (e.g. 0.1), I get a loss of "nan" after a few hundred iterations. However, if I set a lower one (like 0.000001), after tens of thousands of iterations I still have high loss values. The trick was to initiate the training session with a low learning rate, and after a few thousand iterations, I restarted the session from a snapshot with a higher learning rate.

Hope this can be of any use in the future.

lbcastro, 2019-01-29:
Did you do anything to your model after finishing training? I'm getting 20+ fps with yours but just 3-4 with mine, and I only have a few classes.
Thanks!

lbcastro, 2019-01-29:
I wrote a question about how you achieved much better detection FPS than me, but then I remembered that I was using a VGG net to train and I deleted the question.

However, I didn't know about that merge_bn.py script, it will be very helpful. Thanks a lot!

closed 2019-01-30 by lbcastro