PINTO0309/MobileNet-SSD-RealSense #18: Problem compiling OpenCV 3.4.2 with your TBB 2018U2 package on Ubuntu 16.04 AMD64

AusMatt, 2019-02-26:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  Intel NUC5i7RYHR
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  x86_64
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  Ubuntu1604
  
**[Required]** Details of the work you did before the problem occurred:  
I'm following your https://github.com/PINTO0309/MobileNet-SSD-RealSense/README.md for an Intel 5th gen NUC running Ubuntu 16.04.6 LTS 64bit.

Specifically, I have an NCS 1 so I'm using the NCSDK2 and following the instruction under heading:
"Work with RaspberryPi3 (or PC + Ubuntu16.04 / RaspberryPi + Ubuntu Mate)"

At step 7, I found your TBBonamd64x86x64 repository and downloaded: https://github.com/PINTO0309/TBBonamd64x86x64/raw/master/libtbb-dev_2018U2_amd64.deb

I'm then trying to build OpenCV 3.4.2 from source (step 9.1) but get stuck at the compilation stage with an error.

**[Required]** Error message:  
/usr/local/include/tbb/atomic.h:222:34: error: ‘tbb::internal::atomic_impl<T>::my_storage’ has incomplete type
     aligned_storage<T,sizeof(T)> my_storage;
                                  ^
/usr/local/include/tbb/atomic.h:95:8: note: declaration of ‘struct tbb::internal::aligned_storage<long int, 8ul>’
 struct aligned_storage;
        ^
/usr/local/include/tbb/atomic.h: In instantiation of ‘struct tbb::internal::atomic_impl<long unsigned int>’:
/usr/local/include/tbb/atomic.h:338:8:   required from ‘struct tbb::internal::atomic_impl_with_arithmetic<long unsigned int, long unsigned int, char>’
/usr/local/include/tbb/atomic.h:444:1:   required from here

**[Required]** Overview of problems and questions:  
Please help!

AusMatt, 2019-02-26:
I've found a work-around by using "-D__TBB_64BIT_ATOMICS=1" at the cmake stage.
Is this what's required on x86_64 (AMD64)?

PINTO0309, 2019-02-26:
@AusMatt 

In "Ubuntu16.04 + x86_64/AMD64", sufficient performance can be obtained with OpenCV introduced by pip installation.

```bash
$ sudo -H pip3 install pip --upgrade
$ sudo -H pip3 install opencv-python --upgrade
```

In your environment, implementation of "step 7, 8, 9.1, 9.2" is unnecessary.
The procedure I created is a procedure optimized for RaspberryPi3.

AusMatt, 2019-02-27:
@PINTO0309 
I think that I'll still need to install OpenCV somehow. The Python3 opencv-python module is just "Python bindings" to use OpenCV isn't it?

The OpenCV that is provided by Ubuntu 16.04 repositories seems rather old so I've built OpenCV 3.4.5 myself. I'm a bit confused as to how to configure the OpenCV that I'm building. I've based my cmake off what you were doing but have added the extra items: "-D WITH_PROTOBUF=ON -DWITH_GDCM=OFF -DWITH_LAPACK=OFF -DENABLE_CXX11=ON -DBUILD_PROTOBUF=OFF -DPROTOBUF_UPDATE_FILES=OFF -DBUILD_LIBPROTOBUF_FROM_SOURCES=OFF" based on what I was reading here: [https://github.com/opencv/opencv/issues/12429](https://github.com/opencv/opencv/issues/12429) to ensure that the OpenCV highgui module is built using the protobuf v3.5.1 that we installed earlier.

In general I find your instructions confusing as you do not explain why you install each item and why it's required on one platform and not another.

My system is now totally messed up in that Python3 just segfaults if I try to run help ('modules').
I'm not sure what to do now other than reinstall from scratch.

In general I think it would be preferable to use a Python virtual environment rather than the system wide python installation locations. I know that this is more confusing but its easier to recover from errors as you can just delete you whole virtual environment.

PINTO0309, 2019-02-27:
@AusMatt 

I'm sorry. I had provided you with erroneous information.
The following is likely to be helpful.
Please change the version of OpenCV by yourself.
**https://www.learnopencv.com/install-opencv-3-4-4-on-ubuntu-16-04/**

>In general I think it would be preferable to use a Python virtual environment rather than the system wide python installation locations. I know that this is more confusing but its easier to recover from errors as you can just delete you whole virtual environment.

In order to make it a specialized explanation for a particular environment, I dare deducted the explanatory information and made it simple.
I avoid using "Python virtual environment" because it makes it difficult to understand the dependencies of environment variables.

I omit the explanation of Ubuntu x86_64, but I always use the following command.
In the first place, the options that can be used depend on your hardware environment.
For example,  "-D WITH_CUDA=ON" will not work unless CUDA is installed.
It is difficult to describe efficient procedures for all environments.

```
$ sudo apt update && sudo apt upgrade
$ sudo apt install build-essential cmake pkg-config libjpeg-dev libtiff5-dev \
libjasper-dev libavcodec-dev libavformat-dev libswscale-dev \
libv4l-dev libxvidcore-dev libx264-dev libgtk2.0-dev libgtk-3-dev \
libcanberra-gtk* libatlas-base-dev gfortran python2.7-dev python3-dev

$ cd ~
$ wget -O opencv.zip https://github.com/Itseez/opencv/archive/3.4.5.zip
$ unzip opencv.zip;rm opencv.zip
$ wget -O opencv_contrib.zip https://github.com/Itseez/opencv_contrib/archive/3.4.5.zip
$ unzip opencv_contrib.zip;rm opencv_contrib.zip
$ cd ~/opencv-3.4.5/;mkdir build;cd build
$ cmake -D CMAKE_BUILD_TYPE=RELEASE \
        -D CMAKE_INSTALL_PREFIX=/usr/local \
        -D INSTALL_PYTHON_EXAMPLES=OFF \
        -D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib-3.4.5/modules \
        -D BUILD_EXAMPLES=OFF \
        -D PYTHON_DEFAULT_EXECUTABLE=$(which python3) \
        -D INSTALL_PYTHON_EXAMPLES=OFF \
        -D BUILD_opencv_python2=ON \
        -D BUILD_opencv_python3=ON \
        -D WITH_OPENCL=OFF \
        -D WITH_OPENGL=OFF \
        -D WITH_TBB=OFF \
        -D BUILD_TBB=OFF \
        -D WITH_CUDA=OFF \
        -D ENABLE_NEON:BOOL=OFF \
        -D ENABLE_VFPV3=OFF \
        -D WITH_QT=OFF \
        -D BUILD_TESTS=OFF ..
$ make -j8
$ sudo make install
$ sudo ldconfig
```

PINTO0309, 2019-03-01:
@AusMatt 

You can install it below.
**https://pypi.org/project/opencv-python/#history**

```bash
$ sudo -H pip3 install pip --upgrade
$ sudo -H pip3 install opencv-python==3.4.5.20 --upgrade
$ sudo -H pip3 install opencv-contrib-python==3.4.5.20 --upgrade
```

AusMatt, 2019-03-04:
@PINTO0309

Thanks. I'll try this soon.