PINTO0309/MobileNet-SSD-RealSense #20: Failed to load multi NCS2

zree, 2019-03-11:
Hi, I tried to load multi NCS2 on RaspberryPi3B+ and Ubuntu16.04LTS with intel i7-7700k just like what you do, using the IEPlugin function. I actually plugged 2 NCS2, and called the function for 2 times, but it failed with a error message saying NO_DEVICE when loading the second NCS2. Have you encountered the same problem before?  I thought a plugin is an abstract for a device, am I right?

PINTO0309, 2019-03-11:
The API specification seems to have been changed.
I think that you realize that this repository implementation has a difference with the following repositories.
**https://github.com/PINTO0309/OpenVINO-EmotionRecognition.git**

sorry.

zree, 2019-03-13:
> I think that you realize that this repository implementation has a difference with the following repositories.
> **https://github.com/PINTO0309/OpenVINO-EmotionRecognition.git**
Yes! I found this repo on Intel CV forum after I sent this issue. You have done so much valuable work!