PINTO0309/MobileNet-SSD-RealSense #21: Got error message "no module named  mvnc" when running SingleStickSSDwithRealSense.py

MesonHu, 2019-03-17:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name): Raspberry PI3 
  
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):   armv7l
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  Raspbian
  
**[Required]** Details of the work you did before the problem occurred:  
<br>
Finish all steps described in the article.
<br>
**[Required]** Error message:  
<br>
ImportError: No module named 'mvnc'
<br>
**[Required]** Overview of problems and questions:  
<br>
Does OpenVino include mvnc? 
<br>

PINTO0309, 2019-03-17:
>Does OpenVino include mvnc?

No.

SingleStickSSDwithRealSense.py -> for NCSDK
SingleStickSSDwithRealSense_OpenVINO_NCS2.py -> for OpenVINO
SingleStickSSDwithUSBCamera_OpenVINO_NCS2.py -> for OpenVINO

MesonHu, 2019-03-17:
Thanks for reply.
Different error occurs as following:
Traceback (most recent call last):
  File "SingleStickSSDwithRealSense_OpenVINO_NCS2.py", line 31, in <module>
    pipeline.start(config)
RuntimeError: Couldn't resolve requests

ps: Intel RealsenseD435 is attached and Realsense sdk 2.0 is installed.

PINTO0309, 2019-03-17:
Please see the following.
https://github.com/IntelRealSense/librealsense/issues/2473

MesonHu, 2019-03-17:
Thanks PINTO039.
I still got some error messages. I try to obtain the supported models by enumerate_devices, it shows:
    Infrared 1	  1280x720	@ 6Hz	Y8
    Infrared 1	  640x480	@ 30Hz	Y8
    Infrared 1	  640x480	@ 15Hz	Y8
    Infrared 1	  640x480	@ 6Hz	Y8
    Infrared 1	  480x270	@ 60Hz	Y8
    Infrared 1	  480x270	@ 30Hz	Y8
    Infrared 1	  480x270	@ 15Hz	Y8
    Infrared 1	  480x270	@ 6Hz	Y8
    Depth	  1280x720	@ 6Hz	Z16
    Depth	  640x480	@ 30Hz	Z16
    Depth	  640x480	@ 15Hz	Z16
    Depth	  640x480	@ 6Hz	Z16
    Depth	  480x270	@ 60Hz	Z16
    Depth	  480x270	@ 30Hz	Z16
    Depth	  480x270	@ 15Hz	Z16
    Depth	  480x270	@ 6Hz	Z16

Some articles mentioned that Color modes only support in USB3.0. Raspberry Pi 3 only equips USB2.0. However, your demonstration seems is executed on Raspberry 3 Pi. So strange..

PINTO0309, 2019-03-17:
Please see the following.
https://realsense.intel.com/intel-realsense-downloads/#firmware

I do not strongly recommend 5.11.
Use only "Production ready release".
"Development release" is always very unstable.

- Firmware 5.10.6
https://downloadmirror.intel.com/28237/eng/Intel%C2%AE%20RealSense%E2%84%A2D400%20Series%20Signed%20Production%20Firmware%20v5_10_6.zip

- SDK 2.16.5
```console
$ git clone -b v2.16.5 https://github.com/IntelRealSense/librealsense.git
$ cd ~/librealsense
$ git checkout -b v2.16.5
$ mkdir build;cd build
$ cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release
# For RaspberryPi3
$ make -j1
$ sudo make install
```

MesonHu, 2019-03-17:
Really appreciate your kind response.
5.10.6 not works for me. There is still no color mode by running rs-enumerate-devices.  Only 5.10.3 contains color mode. Neither in 5.10.13 and other versions.
But depth mode was disappeared in 5.10.3. Therefore, I have to remark the related codes in  SingleStickSSDwithRealSense_OpenVINO_NCS2.py.
However, both color and depth modes work in MultiStickSSDwithRealSense_OpenVINO_NCS2.py. I will try to discover the differences between these two codes.