PINTO0309/MobileNet-SSD-RealSense #22: Could you share the way of convert MobileNet-ssd to IR?

potoo0, 2019-03-24:
Could you share the way of convert MobileNet-ssd to IR?  
I convert Mobilenet-ssd v2 to IR, but got lots of error. Here is my approach:  
```powershell
python3 .\mo_tf.py `
--input_model E:\ssdlite_mobilenet_v2\frozen_inference_graph.pb `
--model_name ssd_mobilenet_v2 `
--output_dir E:\ssdlite_mobilenet_v2\ `
--data_type FP16 `
--batch 1 `
--tensorflow_object_detection_api_pipeline_config E:\ssdlite_mobilenet_v2\pipeline.config `
--tensorflow_use_custom_operations_config .\extensions\front\tf\ssd_v2_support.json
```
But there are many errors when I use this converted IR model.

PINTO0309, 2019-03-24:
If you are genuinely asking for support, you should write down error messages and environmental information in detail.
Follow the issue template.

potoo0, 2019-03-27:
Sorry about that. Thanks for your time and patience.
**my system**:  
- *device*: Raspberry Pi3B+
- *CPU architecture*: armv7l
- *NCS stick*: NCS v1
- *OS*: Raspbian 9.8
- *python_ver*: 3.5.3
- *tensorflow_ver*: 1.12.0

I using [Tensorflow detection model zoo](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md)'s [ssdlite_mobilenet_v2_coco](http://download.tensorflow.org/models/object_detection/ssdlite_mobilenet_v2_coco_2018_05_09.tar.gz) provided by Tensorflow official. Then convert it to IR model.   
```bash
python ./mo_tf.py \
--input_model ~/ssdlite_mobilenet_v2/frozen_inference_graph.pb \
--model_name ssd_mobilenet_v2 \
--output_dir ~/ssdlite_mobilenet_v2 \
--batch 1 \
--reverse_input_channels \
--tensorflow_object_detection_api_pipeline_config ~/ssdlite_mobilenet_v2/pipeline.config \
--tensorflow_use_custom_operations_config ~/extensions/front/tf/ssd_v2_support.json
```
But when I using your code [SingleStickSSDwithUSBCamera_OpenVINO_NCS2.py](https://github.com/PINTO0309/MobileNet-SSD-RealSense/blob/master/SingleStickSSDwithUSBCamera_OpenVINO_NCS2.py) to test model if it works, 
, I never got any detections. I found the out of cv2.dnn is always zero except first element is -1. 
I follow this issue [MobileNet-SSD + RPi + NCS2](https://github.com/PINTO0309/MobileNet-SSD-RealSense/issues/16), but I have no idea about original model's info, so tried different *data_type*, none of them work. I don't know where to find the normalization's param of COCO training set, but I'm working on it.

Could you help me figure it out? Is the way to convert the model incorrect?

potoo0, 2019-04-02:
I use this param of normalization, it works fine: `mean=128, std=1.0`.

closed 2019-04-02 by potoo0

potoo0, 2019-04-04:
Here is my summary:
- pre-trained model provide by tensorflow: [Object Detection Models Zoo](https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/detection_model_zoo.md)
- If the model above doesn't specify the normalization's param, then `cv2.dnn.blobFromImage` shouldn't pass in `scalefactor` and `mean`. 
- I test *ssdlite_mobilenet_v2_coco*, there is no need to `reverse_input_channels` when convert model to IR. And `cv2.dnn.blobFromImage` shouldn't set `swapRB=False`.

PINTO0309, 2019-04-04:
Thank you for providing very useful information.
It will be a very good help for other engineers.

potoo0, 2019-04-04:
My pleasure.