PINTO0309/MobileNet-SSD-RealSense #23: Raspberry pi3b/3b+ + RealSense D435 + no NCS or NCS1/2

isac1120, 2019-04-21:
**[Required]** Your device (RaspberryPi3, LaptopPC, or other device name):  
  Raspberry Pi3b / Raspberry pi3b+
**[Required]** Your device's CPU architecture (armv7l, x86_64, or other architecture name):  
  
**[Required]** Your OS (Raspbian, Ubuntu1604, or other os name):  
  Raspbian / Ubuntu 16.04
**[Required]** Details of the work you did before the problem occurred:  
<br>i've already tried your project PINTO0309/MobileNet-SSDLite-RealSense-TF . Thank you for your project. And now,I'm considering using NCS v1 or v2 to improve performance of raspberry pi3b/3b+ with realsense.
<br>
<br>
**[Required]** Error message:  
<br>
<br>
<br>
**[Required]** Overview of problems and questions:  
<br>I wonder how much fps or perfermance difference between no NCS  and using NCS v1 and NCS v2. Thank you.
<br>
<br>

PINTO0309, 2019-04-22:
There is a difference depending on the resolution, so it can not be said in general.
I listed the URL of the video in the Readme, did you see it?
RealSense can not be set lower than 640x480, so it is a bit slower than 320x240.

- NCS2 + RealSense D435, 16 FPS 640x480

- NCS2 + USBCam, 24 FPS 320x240
https://www.youtube.com/watch?v=Xj2rw_5GwlI

- No NCS2 + RealSense D435, 0.5 FPS 640x480
https://www.youtube.com/watch?v=o5Jh5U9jPFQ

isac1120, 2019-04-23:
Thank you for your help :)

isac1120, 2022-08-16:
Again hello, Pinto0309! Thanks to your help, I successfully completed my graduation project. Without your contribution, my team could't finished this project. So, I wanted to show you our project. ( It is unveiled to general people. ) At that time, I couldn't find your e-mail, so I gave up to contact you. So, I write letter on the github now cause I remember the project again now. Thank you! 
=> https://youtu.be/eZ5zDiYjGCc

closed 2022-08-16 by isac1120