PINTO0309/MobileNet-SSD-RealSense #24: Could you make guide to make merge model of object detection and face detection

Qlinkcopter, 2019-04-25:
I'm in learning to use openvino and movidius1 
your github example is great!!

And if you pleased, i would like to know how you merge object detection and face detection before convert to IR 

Thank you

PINTO0309, 2019-05-11:
I have already released the program, so I do not know what you want to know.
It only uses two different models.
btw, OpenVINO and NCS1 can not be combined.

- NCS1 (movidius1) + NCSDK
**https://github.com/PINTO0309/MobileNet-SSD-RealSense/blob/master/MultiStickSSDwithRealSense.py**

- NCS2 + OpenVINO, EmotionRecognition + FaceDetection 
**https://github.com/PINTO0309/OpenVINO-EmotionRecognition.git**

Qlinkwisp, 2019-05-22:
may be i misunderstood 

let's me explain more

i think mobilessd has 21 labels things from background to tvmonitor that can give 21 possible inference label answer

and you also can give answer of [background] and [face] 

im not sure how to do this

Please point the way i can learn more

PINTO0309, 2019-05-22:
@Qlinkwisp 
Pascal VOC labels example
```
[Background]
Person
Car
Bicycle
Bus
Motorbike
Train
Aeroplane
Chair
Bottle
Dining Table
Potted Plant
TV/Monitor
Sofa
Bird
Cat
Cow
Dog
Horse
Sheep
```
If you want to generate a "Face" trained model, do learning with two labels as follows:
```
[Background]
Face
```
Or just carry out learning with 22 labels as shown below.
```
[Background]
Person
Car
Bicycle
Bus
Motorbike
Train
Aeroplane
Chair
Bottle
Dining Table
Potted Plant
TV/Monitor
Sofa
Bird
Cat
Cow
Dog
Horse
Sheep
Face
```