PINTO0309/MobileNet-SSD-RealSense #26: Got error message "USB Failure. Code: Error opening device"

mumuwo, 2019-05-09:
**[Required]** Your device (RaspberryPi3):  
  
**[Required]** Your device's CPU architecture (armv7l):  
  
**[Required]** Your OS (Raspbian):  
  
**[Required]** Details of the work you did before the problem occurred:  
<br>I followed your steps to install ncsdk on the Raspberry Pi. An error occurred while executing make examples
**[Required]** Error message:  
# Network Input tensors ['data#24']
# Network Output tensors ['prob#38']
Blob generated
W: [         0] ncDeviceOpen:528	ncDeviceOpen() XLinkBootRemote returned error 3

[Error 7] Toolkit Error: USB Failure. Code: Error opening device
Makefile:64: recipe for target 'profile' failed
make[3]: *** [profile] Error 255
make[3]: Leaving directory '/home/pi/ncsdk/examples/caffe/AlexNet'
Makefile:12: recipe for target 'AlexNet' failed
make[2]: *** [AlexNet] Error 2
make[2]: Leaving directory '/home/pi/ncsdk/examples/caffe'
Makefile:12: recipe for target 'caffe/.' failed
make[1]: *** [caffe/.] Error 2
make[1]: Leaving directory '/home/pi/ncsdk/examples'
Makefile:57: recipe for target 'examples' failed
make: *** [examples] Error 2
**[Required]** Overview of problems and questions:  
Unable to find device

PINTO0309, 2019-05-11:
Please check the following.
1. Using NCS1 (Movidius Neural Compute Stick). NCS2 is not supported.
2. Power supply voltage must be sufficient
3. Not using low quality USB Hub
4. Do not use virtualization software such as VirtualBox

mumuwo, 2019-05-11:
Thank you very much for your advice.
I don't know that NCSDk does not support NCS2.
Now, I have installed openvino and it runs successfully, thank you for your help.

closed 2019-05-13 by mumuwo